tu-darmstadt-ros-pkg/hector_gazebo #18: Level not equal to 1 in the SensorModel dynamic reconfigure server

jim-rothrock, 2015-02-23:
I have found that in SensorModel_<T>::dynamicReconfigureCallback() (sensor_model.h), "level" is 4294967295 when the callback is called during gazebo_ros_imu initialization. Therefore, the "if (level == 1)" block isn't executed, and the "else" block initializes my simulated IMU's acceleration offset to an unexpected value ("rosparam get /imu/data/accel/offset" displays 5.046653718333332, when the offset should be zero). Should "if (level == 1)" actually be "if (level & 1)"?

meyerj, 2015-02-23:
> I have found that in SensorModel_::dynamicReconfigureCallback() (sensor_model.h), "level" is 4294967295 when the callback is called during gazebo_ros_imu initialization.

Correct, or more precise: `~0`.

> Therefore, the "if (level == 1)" block isn't executed, and the "else" block initializes my simulated IMU's acceleration offset to an unexpected value ("rosparam get /imu/data/accel/offset" displays [...]

No. The callback was written like that because I did neither want to initialize the sensor model with the default values from the dynamic_reconfigure config file nor from the ROS parameter server, as it would be the case with a "standard" callback with `if (level & 1) ...`. Instead, the initial callback populates the config struct with the settings which have been loaded from the SDF sensor config before. These values set in the callback are also stored in the parameter server and published to dynamic_reconfigure clients during initialization of the dynamic_reconfigure server. This enables usage of dynamic_reconfigure for dynamic updates without breaking existing robot models that set the sensor error parameters from the robot description.

I do not see why this approach would be invalid and cannot reproduce the issue with unexpected values. I admit that for the `gazebo::math::Vector3` instantiation of the SensorModel class like in the IMU acceleration and rate model the conversion to scalar values for the config struct is not very elaborate: It simply takes the average of x, y, z, based on the assumption that the parameters are the same for all three axes in most cases. It is not possible to set them separately with dynamic_reconfigure.

5.046653718333332 is indeed an unrealistic high offset. The SensorModel class initializes the ´offset´ member variable to its default value in the [constructor](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/blob/indigo-devel/hector_gazebo_plugins/include/hector_gazebo_plugins/sensor_model.h#L83), which is 0.0 in case of a [gazebo::math::Vector3](https://bitbucket.org/osrf/gazebo/src/667b26336ea3d56340b6943f7efc043ed11d6db3/gazebo/math/Vector3.cc?at=default#cl-35). Could you post the section of the robot description XML where the sensor plugin loaded? Only if the `<accelOffset>` or `<rpyOffset>` (which is converted into an accelerometer offset internally) tags are present and have non-zero values I could explain a non-zero value of `/imu/data/accel/offset` parameter on the parameter server.

jim-rothrock, 2015-02-23:
> Could you post the section of the robot description XML where the sensor plugin loaded?

Here is the section from the robot's URDF file:

    <gazebo>
      <plugin name="imu" filename="libhector_gazebo_ros_imu.so">
        <bodyName>imu_frame</bodyName>
        <topicName>imu/data</topicName>
        <serviceName>imu/calibrate</serviceName>
        <updateRate>100.0</updateRate>
      </plugin>
    </gazebo>

Here the corresponding section of the SDF that is generated from the URDF:

    <plugin name='imu' filename='libhector_gazebo_ros_imu.so'>
      <topicName>imu/data</topicName>
      <serviceName>imu/calibrate</serviceName>
      <updateRate>100.0</updateRate>
      <bodyName>base_link</bodyName>
      <xyzOffset>0.3823 -0.0039 -0.3007</xyzOffset>
      <rpyOffset>3.11419 1.57032 -3.14159</rpyOffset>
    </plugin>

It appears that the problem might be related to URDF-to-SDF conversion, which inserts an \<rpyOffset\> element with nonzero values.

meyerj, 2015-02-23:
> It appears that the problem might be related to URDF-to-SDF conversion, which inserts an <rpyOffset> element with nonzero values.

I agree. Obviously the SDF conversion assumes that all sensor plugins interpret the `<xyzOffset>` and `<rpyOffset>` parameters in the same way as an additional sensor link which is connected with a static joint to the real parent frame. I was not aware that this is a requirement. `hector_gazebo_ros_imu` interprets the roll and pitch part of `<rpyOffset>` as an orientation offset caused by a (small) accelerometer bias. This can never work for rpyOffsets as big as in your SDF example.

I will prepare a small patch that corrects this. Unfortunately version 0.3.5 of hector_gazebo was just released a few hours ago. Perhaps I can re-release a fixed version before the next sync.

jim-rothrock, 2015-02-24:
> Perhaps I can re-release a fixed version before the next sync.

There's no rush. I worked around the problem by commenting out the code in gazebo_ros_imu.cpp that is executed when an rpyOffset element is found.

commit 74b21b74 (2015-03-21): hector_gazebo_plugins/hector_gazebo_ros_imu: interpret parameters xyzOffset and rpyOffset as pure mounting offsets and not as induced by accelerometer bias (fix #18)

Obviously the SDF conversion assumes that all sensor plugins interpret the `<xyzOffset>` and `<rpyOffset>` parameters in the same way as an
additional sensor link which is connected with a static joint to the real parent frame. I was not aware that this is a requirement.
hector_gazebo_ros_imu interpreted the roll and pitch part of `<rpyOffset>` as an orientation offset caused by a (small) accelerometer bias.

This patch completel [... cut]
closed 2015-03-21 by meyerj