tu-darmstadt-ros-pkg/hector_gazebo #23: rpyOffset does not affect angular rates

bgoldfai, 2015-12-09:
I use hector_gazebo IMU in my robot URDF and to match the orientation of the IMU on our robot I roll it by 180 degrees (z-down, x-forward) using `<rpyOffset>3.14159 0 0</rpyOffset>`. When I visualize the linear_acceleration components of the published imu message, all values are transformed correctly, but the angular_rate values don't appear to be transformed (they are the same as when the roll was 0). I tested this in Gazebo using a joystick to drive our robot around and visualized the imu data with rqt_plot.

closed 2016-06-24 by meyerj

meyerj, 2016-06-24:
Fixed in d5fcb4a. The patch will be released in indigo (0.3.8), jade (0.4.1) and kinetic (0.5.0).
Thanks for reporting.

bgoldfai, 2016-06-27:
Great, thanks for the fix.

commit d5fcb4a4 (2016-06-24): hector_gazebo_plugins: fixed angular rate output with non-zero rpyOffset parameter (fix #23)