tu-darmstadt-ros-pkg/hector_gazebo #26: How to add RFIDs or Wireless Transceivers  as a plugin the plugin?

libing64, 2016-03-03:
This is the sensor list in gazebo2.2  (/usr/include/gazebo-2.2/gazebo/sensors), I want to add the rfid sensor and to wireless transceiver as a plugin to the platform, but I am not sure how to do that.
Can anyone give some advice?

- CameraSensor.hh
- ContactSensor.hh
- DepthCameraSensor.hh
- ForceTorqueSensor.hh
- GpsSensor.hh
- GpuRaySensor.hh
- ImuSensor.hh
- MultiCameraSensor.hh
- Noise.hh
- RaySensor.hh
- RFIDSensor.hh
- RFIDTag.hh
- SensorFactory.hh
- Sensor.hh
- SensorManager.hh
- sensors.hh
- SensorsIface.hh
- SensorTypes.hh
- SonarSensor.hh
- WirelessReceiver.hh
- WirelessTransceiver.hh
- WirelessTransmitter.hh

libing64, 2016-03-07:
I have successfully added the wireless transceiver plugin to the simulation platform, I just search all the models with the name of "transmitter" and then compute the relative pose, so the distance and angle of arrival can be used as the measurement of the wireless sensor.

closed 2016-03-07 by libing64

nicolaerosia, 2016-03-07:
Interesting! I was wondering if you could share snippets of code for this.
Best Regards

libing64, 2016-03-07:
@nicolaerosia 
I upload the code to my github
https://github.com/libing64/hector_gazebo

plumewind, 2020-08-13:
hello,

I use the the wireless transceiver plugin, when I catkin_make get this(on gazebo9 ubuntu18.04 melodic):
 
`/hector_gazebo_plugins/src/gazebo_ros_wireless_receiver.cpp:237:54: error: conversion from ‘ignition::math::v4::Vector3<double>’ to non-scalar type ‘ignition::math::v4::Pose3d {aka ignition::math::v4::Pose3<double>}’ requested
       ignition::math::Pose3d dir  = AoA_sensor_model_(relative_pose.Pos(), dt);
                                     ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_wireless_receiver.dir/build.make:62: recipe for target 'hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_wireless_receiver.dir/src/gazebo_ros_wireless_receiver.cpp.o' failed
`
can you help me? thanks