tu-darmstadt-ros-pkg/hector_gazebo #32: Release Kinetic branch

davetcoleman, 2017-04-12:
The [change](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/commit/271695d90e9115d5ab53ecb5446525790f10c313) for depending on Gazebo is not released yet, and so I think that is the reason ``gazebo_ros_pkgs`` builds are [failing](http://build.osrfoundation.org/job/ros_gazebo_pkgs-ci-pr_regression_any-xenial-amd64/26/console) (it uses this repo as a test). @j-rivero might have more insight on this, also

Also make ``kinetic-devel`` the default branch for Github?

meyerj, 2017-08-09:
I think this has been obsoleted by https://github.com/ros-simulation/gazebo_ros_pkgs/pull/571, no? At least I cannot reproduce the issue locally with `ros-kinetic-gazebo-dev` version `2.5.13-0xenial-20170624-093007-0800` installed. Can someone confirm this?

What triggers the [ros_gazebo_pkgs-ci-pr_regression_any-xenial-amd64](https://build.osrfoundation.org/job/ros_gazebo_pkgs-ci-pr_regression_any-xenial-amd64/) job? Or has it been replaced by [ros_gazebo_pkgs-ci-pr_regression_any_kinetic-xenial-amd64](https://build.osrfoundation.org/job/ros_gazebo_pkgs-ci-pr_regression_any_kinetic-xenial-amd64/), which apparently builds successfully?

j-rivero, 2017-08-09:
> What triggers the ros_gazebo_pkgs-ci-pr_regression_any-xenial-amd64 job? Or has it been replaced by ros_gazebo_pkgs-ci-pr_regression_any_kinetic-xenial-amd64, which apparently builds successfully?

It was replaced yes and it builds fine so I don't think that you need to do any action at this moment. Thanks.

closed 2017-08-09 by meyerj