tu-darmstadt-ros-pkg/hector_gazebo #34: hector_gazebo failing to build in Kinetic

tfoote, 2017-04-22:
This appears to be a missing dependency on pthread. Which was previously masked by gazebo_ros and gazebo_plugins

It was released 4 days ago with the dependencies cleaned up: https://github.com/ros/rosdistro/pull/14563 warning that downstream may have an issue with pthreads.

tfoote, 2017-04-22:
There's a known issue in catkin
ros/catkin#856

meyerj, 2017-04-24:
I don't fully understand what we can do to fix this:
- Wait until https://github.com/ros/catkin/issues/856 has been fixed?
- Do not `find_package(... gazebo_plugins)` in [hector_gazebo_thermal_camera/CMakeLists.txt:7](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/blob/indigo-devel/hector_gazebo_thermal_camera/CMakeLists.txt#L7)? But we include a header provided by that package.
- Modify the CMake code to find Gazebo or `gazebo_plugins`, without relying on catkin?

What is the recommended solution?

tfoote, 2017-04-24:
I'm not sure there's discussion at: https://github.com/yujinrobot/kobuki_desktop/issues/50

@j-rivero FYI

tfoote, 2017-04-26:
Fixed by gazebo_ros_pkgs rollback. Closing.

closed 2017-04-26 by tfoote