tu-darmstadt-ros-pkg/hector_gazebo #42: Incorrect link name/position combinations in GPS plugin

tcassell94, 2017-11-27:
I haven't fully investigated the extent of this issue, but here's what I've found in our situation. We use both the IMU and GPS plugins, which are attached to the top of a pivoting joint in the URDF through static joints (imu_link -> top and gps_link -> top both static). The GPS plugin reports the altitude as the altitude of the 'top' link (the parent of gps_link), rather than the altitude of gps_link itself. However, in the message output by the plugin, the 'frame_id' field is still 'gps_link'. I haven't investigated if it would do the same for lat/lon if 'gps_link' was xy offset from 'top'.

Guillaumebeuzeboc, 2019-04-17:
Hello, I have the same issue. In my case, it's with lat/long.
My setup is base_link->gps_link. Although I attach the bodyName to gps_link the gps return to a position of base_link.
An easy way to check it is to spawn a robot in gazebo with a gps_link not centered on base_link and rotate the robot inside gazebo (the rotation will happen around the base_link, so base_link won't move in x,y while the gps_link will describe a circle) and you will see that the lat/long is not changing (like if the gps was locating the base_link and not the gps_link)

plumewind, 2020-08-17:
hello ，

I have got the problem，([issue 74](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/issues/74))

I did not understand this either:
If your link doesn't have any inertia it won't be created in gazebo as well .

Is that mean, it will ok by using a link with inertial?

Thanks a lot.

Martin-Oehler, 2020-08-18:
This is a known problem with the URDF to SDF conversion, because historically ODE couldn't handle fixed joints. Now, you can preserve fixed joints with
```
<gazebo reference="JOINT_NAME_HERE">
    <preserveFixedJoint>true</preserveFixedJoint>
</gazebo>
```

closed 2020-08-18 by Martin-Oehler