tu-darmstadt-ros-pkg/hector_gazebo #50: IMU plugin

tongtybj, 2018-05-01:
I am considering to use imu plugin in gazebo.
What is the main difference between hector imu plugin and [GazeboRosImuSensor](http://gazebosim.org/tutorials?tut=ros_gzplugins)?

meyerj, 2018-05-02:
The `GazeboRosImuSensor` plugin was only added/merged recently (https://github.com/ros-simulation/gazebo_ros_pkgs/pull/363), while the IMU plugin in hector_gazebo dates back to 2012. Back then there was no proper IMU plugin in [ros-gazebo-pkgs](https://github.com/ros-simulation/gazebo_ros_pkgs), that

- includes gravity in the acceleration output, like real accelerometers would do,
- can simulate sensor drift and a non-zero initial bias,
- also applies an error to the orientation output, which is correlated to the simulated accelerometer drift, and
- is reconfigurable through dynamic_reconfigure and services at run-time to adjust the error parameters or to "calibrate" the IMU (reset the bias state to zero). 

The approach of `GazeboRosImuSensor`, that uses Gazebo's `ImuSensor` internally, is definitely cleaner and should be preferred for new projects. But I have not compared the output of both plugins in detail yet. I think the last two "features" are still unique to `hector_gazebo_ros_imu`.

Another IMU plugin I am aware of is provided by the [rotors_simulator](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_imu_plugin.cpp).

tongtybj, 2018-05-02:
@meyerj 

Thank you for your kind explanation! I also agree that the last two features are important. Hope GazeboRosImuSensor can also implement these features.