tu-darmstadt-ros-pkg/hector_gazebo #68: Noetic release

Timple, 2020-05-25:
Would you mind releasing this package also for the newest distibution ROS Noetic? :)

StefanFabian, 2020-05-25:
Sure but it will take a few weeks until I can get around to setting up a noetic installation.

rgmann, 2021-02-01:
Bump. I would also be interested in a Noetic release.

StefanFabian, 2021-02-02:
Thank you for the bump. I thought it was already released but apparently something isn't working with the build process.
The release repo doesn't have a tag that the build farm is looking for. I'll look into that.

rgmann, 2021-02-03:
@StefanFabian Thank you!

StefanFabian, 2021-02-09:
Packages should now be available.

closed 2021-02-09 by StefanFabian

NedaRH, 2023-05-29:
Could you please provide the link of Noetic release of this package? I couldn't find it.

StefanFabian, 2023-05-30:
The default branch is always the most up-to-date one for our packages.
There were no changes that weren't compatible with melodic, hence, we did not create a new branch.
As of the noetic release, the ros build farm distributes the packages. You can install them separately or all of them with the meta package `ros-noetic-hector-gazebo`.