tu-darmstadt-ros-pkg/hector_gazebo #74: Can not get the link of sensor in robot by the mode type plugin

plumewind, 2020-08-14:
hello,

When I use the plugin(gps, imu,etc are tested), I can not get the link of sensor, it report:link is not exist.

As a exmaple, a imu plugin, I put it on a robot, I wrote the urdf.xacro(a block):

`
......
  <xacro:prius_stack/>

  <xacro:sensor_imu parent="chassis">
     <origin xyz="0 0 0.8" rpy="0 0 -1.57"/>
  </xacro:sensor_imu> 
......
`
some links can be:base_link->chassis->imu_link

In the model type plugin(gazebo_ros_imu.cpp), I check the links like this:
`
std::vector<physics::LinkPtr> links = _model->GetLinks(); 
for (int i = 0; i < links.size(); ++i) 
{ 
ROS_INFO("Link[%d]: %s", i, links[i]->GetName().c_str());
 }
`
But no imu_link, so the link->getWorldPose() is wrong. And, the chassis is exist, getWorldPose() is right.
But, in rqt_tf_tree, the imu_link is exist, the pub rate of imu_link is 1000, the pub rate of chassis is 10.

what should i do? Thanks a lot.

Martin-Oehler, 2020-08-18:
Hello,
when converting the URDF to SDF, fixed joints will be removed from the model. The attached links will instead be added to the parent link and, therefore, won't exist when trying to add the plugin. Links without inertia are ignored altogether. 
To fix your problem, you have to redesign your urdf and attach the sensor to a link, that won't be removed. Alternatively, you can preserve the joint with
```
    <gazebo reference="JOINT_NAME_HERE">
      <preserveFixedJoint>true</preserveFixedJoint>
    </gazebo>
```

plumewind, 2020-08-18:
Thanks， I will have a try !

plumewind, 2020-08-18:
Thanks a lot. 
I fixed it !
cool !
`    <gazebo reference="gps_joint">
        <preserveFixedJoint>true</preserveFixedJoint>
    </gazebo>`

closed 2020-08-18 by Martin-Oehler

plumewind, 2020-08-18:
And I find a warning:

[ WARN] [1597752408.295953452]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.

The root link with an inertia, the robot can be saw in gazebo. But if I wrote:
<link name="base_link"/>
There is no robot in gazebo. 

Does it is same to this?

Martin-Oehler, 2020-08-18:
The root link will never be removed. An inertia tag is not required and, apparently, not supported by KDL. You can move the inertia tag to a chassis link and connect it with a fixed link.

plumewind, 2020-08-19:
Thanks, I fixed it.