tu-darmstadt-ros-pkg/hector_gazebo #78: Explanatory README

TadielloM, 2020-08-31:
Hello,
I am trying to install the package and run the sensor model in some kind of robot but I am not able to figure out how to add the sensor as a model plugin in an SDF model.
In the README there is nothing that explains how to install and how to run it properly, neither an example folder to get some suggestion where to start. I think that a more explanatory README with an installation and example section can easily remove a lot of problems to new developer trying this package.

StefanFabian, 2020-08-31:
I absolutely agree. However, the original authors are not available anymore and we only maintain it to ensure that it remains compatible with new ROS and Gazebo versions.
We would gladly accept contributions with examples or documentation.

TadielloM, 2020-09-05:
I can try to run it and write something a little bit more clear in the readme. But from what I understand there are different plugins in the same repo. I mainly need the thermal camera.  If I get how to use it I can write a readme in the ```hector_gazebo_thermal_camera``` folder. 

However, I mainly use SDF file format and I don't get how to include the package and the model of the camera in the robot model.