tu-darmstadt-ros-pkg/hector_gazebo #82: Missing #include <gazebo/math/Vector3.hh>

JohnVorwald, 2021-01-02:
I new.  Appears I'm using gazebo 9 which no longer has Vector3.hh, and is called by gazebo_ros_magnetic.h.

ref: https://answers.gazebosim.org//question/21872/vector3hh-not-found-on-gazebo9/
Vector3 source code is here: https://github.com/arpg/Gazebo/tree/master/gazebo/math
These math functions are pretty trivial.
The proposed replacement is ignition math
Ignition math online source is here: https://ignitionrobotics.org/api/math/6.1/Vector3_8hh_source.html
Local source code is: /usr/include/ignition/math4/ignition/math/Vector3.hh

Has anyone addressed this issue, which appears to be due to update from Gazebo 8 to 9.  Maybe I should just install Gazebo 8.
I'm using Ubuntu 18.04.

[  2%] Building CXX object hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/src/gazebo_ros_magnetic.cpp.o
In file included from /home/johnvorwald/catkin_ws/src/hector_gazebo/hector_gazebo_plugins/include/hector_gazebo_plugins/gazebo_ros_magnetic.h:36:0,
                 from /home/johnvorwald/catkin_ws/src/hector_gazebo/hector_gazebo_plugins/src/gazebo_ros_magnetic.cpp:29:
/home/johnvorwald/catkin_ws/src/hector_gazebo/hector_gazebo_plugins/include/hector_gazebo_plugins/sensor_model.h:33:10: fatal error: gazebo/math/Vector3.hh: No such file or directory
 #include <gazebo/math/Vector3.hh>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/build.make:62: recipe for target 'hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/src/gazebo_ros_magnetic.cpp.o' failed
make[2]: *** [hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/src/gazebo_ros_magnetic.cpp.o] Error 1
CMakeFiles/Makefile2:5681: recipe for target 'hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/all' failed
make[1]: *** [hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

StefanFabian, 2021-01-04:
Which branch did you check out?
Melodic-devel compiles for me using Ubuntu 18.04 with ROS Melodic and Gazebo 9.0.

JohnVorwald, 2021-01-04:
Interesting, I'm also on branch Melodic-devel, Ubuntu 18.04, ROS melodic, Gazebo 9.0.  But, I'm on a Xavier.  Here are my commands to reproduce, from fresh flash.  These commands are followed by commands showing versions.  So, I probably have an error in my commands to setup hector.

# These first commands are to setup / run turtlebot which I installed before hector.

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src
git clone -b melodic-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git
cd turtlebot3_simulations
git checkout melodic-devel
cd ~/catkin_ws && catkin_make
# Add line “export TURTLEBOT3_MODEL=burger” to ~.bashrc
gedit ~/.bashrc
source ~/.bashrc
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
cd ~/catkin_ws && catkin_make
cd ~

export ROS_PACKAGE_PATH=/opt/ros/melodic/share:/home/johnvorwald/catkin_ws/src/turtlebot3:/home/johnvorwald/catkin_ws/src/turtlebot3_simulations

roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch

# try running a uav sim, 

sudo apt update
sudo apt install python-wstool

# Source code
mkdir ~/catkin_ws/src/hector_quadrotor_tutorial
cd ~/catkin_ws/src/hector_quadrotor_tutorial
wstool init src https://raw.github.com/tu-darmstadt-ros-pkg/hector_quadrotor/hydro-devel/tutorials.rosinstall

# Dependencies
sudo apt-get install qt4-default
sudo apt-get install ros-melodic-geographic-msgs
sudo apt install ros-melodic-driver-base

# Build 
catkin_make
source devel/setup.bash

[  3%] Building CXX object hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/src/gazebo_ros_magnetic.cpp.o
In file included from /home/johnvorwald/catkin_ws/src/hector_quadrotor_tutorial/src/hector_gazebo/hector_gazebo_plugins/include/hector_gazebo_plugins/gazebo_ros_magnetic.h:36:0,
                 from /home/johnvorwald/catkin_ws/src/hector_quadrotor_tutorial/src/hector_gazebo/hector_gazebo_plugins/src/gazebo_ros_magnetic.cpp:29:
/home/johnvorwald/catkin_ws/src/hector_quadrotor_tutorial/src/hector_gazebo/hector_gazebo_plugins/include/hector_gazebo_plugins/sensor_model.h:33:10: fatal error: gazebo/math/Vector3.hh: No such file or directory
 #include <gazebo/math/Vector3.hh>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/build.make:62: recipe for target &apos;hector_gazebo/hector_gazebo_plugins/CMakeFiles/hector_gazebo_ros_magnetic.dir/src/gazebo_ros_magnetic.cpp.o&apos; failed
Appears Gazebo Vector3.hh has been depreciated
ref: https://answers.gazebosim.org//question/21872/vector3hh-not-found-on-gazebo9/
Source code is here: https://github.com/arpg/Gazebo/tree/master/gazebo/math
The functions are pretty trivial.
The proposed replacement is ignition math
Ignition math online source is here: https://ignitionrobotics.org/api/math/6.1/Vector3_8hh_source.html
Local source code is: /usr/include/ignition/math4/ignition/math/Vector3.hh

Here are commands showing versions

~/catkin_ws/src/hector_gazebo$ git status
On branch melodic-devel
Your branch is up to date with 'origin/melodic-devel'.

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic

$ rosversion -d
melodic

$ gazebo --version
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

JohnVorwald, 2021-01-05:
It appears I initially installed ROS incorrectly, following the commands above.  I followed the directions in the book ROS Robotics Projects, and now hector compiles without error.  In case someone has this problems, the following commands corrected the environment.

## source.list and keys

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'

sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116 

sudo apt-get update
sudo apt-get install ros-melodic-desktop-full 

sudo rosdep init
rosdep update

echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc 
source ~/.bashrc 

sudo apt-get install python-rosinstall

roscore

closed 2021-01-05 by JohnVorwald