tu-darmstadt-ros-pkg/hector_gazebo #85: libgazebo_ros_gps : failed to load plugin

Harry-Rogers, 2021-03-01:
Hi,

I have been trying to add this package to my gazebo simulation but have came across this error below when loading the plugin. I'm not sure how to fix this?

[Err] [Plugin.hh:178] Failed to load plugin: undefined symbol: _ZN19dynamic_reconfigure14__init_mutex__E

StefanFabian, 2021-03-01:
I currently don't have the time to look into this but this means it is looking for a function that is in the plugin library or one of its linked libraries.
This could either mean that the library this symbol is in was not linked against the plugin but as far as I could see it should be or that the library changed and the new version may not have this symbol and is therefore not compatible.
In the latter case, the plugin has to be rebuilt.
Also, make sure your packages are all up to date.

Harry-Rogers, 2021-03-01:
I think I'm missing the GNSSConfig.h file. I can't seem to find it on the repository.

StefanFabian, 2021-03-01:
That is already in the CMakeLists.txt of hector_gazebo_plugins so I don't understand why you'd have to add this.

https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/blob/8e2a2a4664330387b68d31f42d0c4de290d33e80/hector_gazebo_plugins/CMakeLists.txt#L7

and

https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/blob/8e2a2a4664330387b68d31f42d0c4de290d33e80/hector_gazebo_plugins/CMakeLists.txt#L44-L48

You also need an `add_dependencies` to make sure it is generated before the target is built:

https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/blob/8e2a2a4664330387b68d31f42d0c4de290d33e80/hector_gazebo_plugins/CMakeLists.txt#L88-L90

closed 2021-03-01 by StefanFabian