tu-darmstadt-ros-pkg/hector_gazebo #87: apt install fails on noetic

Timple, 2021-05-07:
Can be reproduced by this minimal dockerfile:

```Dockerfile
FROM ros:noetic-robot
RUN sed -i 's|ros/ubuntu|ros-testing/ubuntu|g' /etc/apt/sources.list.d/ros1-latest.list
RUN apt-get update -qq && apt-get install -qqy ros-noetic-hector-gazebo-plugins
```

StefanFabian, 2021-05-07:
I've noticed this in a different context but this is not caused by us.
There are some shenanigans going on with the gazebo11 packages.
Seems like a version conflict in one of gazebo's dependencies.
(If this is the same problem, you're encountering.)
I've done the command in an interactive docker container:

```
$ sudo apt install ros-noetic-hector-gazebo-plugins
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-noetic-hector-gazebo-plugins : Depends: libgazebo11 (>= 11.3.0) but it is not going to be installed
                                    Depends: gazebo11 but it is not going to be installed
                                    Depends: ros-noetic-gazebo-ros but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
```
$ sudo apt install gazebo11
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo11 : Depends: libgazebo11 (= 11.5.0-1~focal) but it is not going to be installed
            Recommends: gazebo11-plugin-base but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
```
$ sudo apt install libgazebo11
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgazebo11 : Depends: libignition-common3 (>= 3.12.0) but 3.11.1-1~focal is to be installed
               Depends: libignition-math6 (>= 6.8.0) but 6.7.0-1~focal is to be installed
E: Unable to correct problems, you have held broken packages.
```

Timple, 2021-05-07:
Aha, maybe related to this then: https://github.com/ros/rosdistro/issues/27851

StefanFabian, 2021-05-07:
No, that issue only results in me getting daily emails from the build farm :(
It appears that the gazebo packages were built against newer versions of the libignition libraries than are available in the ubuntu repositories.
It needs >=3.12 but only 3.11 is available.

Timple, 2021-05-07:
So the rosdistro repo would be the best place for this issue?

StefanFabian, 2021-05-07:
There seems to be an issue in the gazebo repository already osrf/gazebo#2995

Timple, 2021-05-08:
Verified to be working again