tu-darmstadt-ros-pkg/hector_gazebo #97: Cannot run the libhector_gazebo_ros_gps.so plugin

austin-InDro, 2022-08-03:
I attemped to compile the foxy-devel branch Hector_Gazebo system on my machine locally since there is not a binary package available for ROS2 yet. I was faced with some CMake issues which I resolved enough to compile, but I fear something else linking wise may be missing since the plugin cannot load into Gazebo properly. 

I've added the `workspace/build/hector_gazebo_plugins` to the `$GAZEBO_PLUGIN_PATH`

The following is some additional information

System: ROS Humble

## Error Output
```bash
gazebo-11.10.2 --verbose -s libhector_gazebo_ros_gps.so /home/austin/Documents/2ws/install/sim_bot_description/share/sim_bot_description/world/my_world.sdf
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.68.139
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [gazebo_shared.cc:46] System is attempting to load a plugin, but detected an incorrect plugin type. Plugin filename[libhector_gazebo_ros_gps.so].
```

## CMake File for Hector_gazebo_plugin
```CMake
cmake_minimum_required(VERSION 3.5)
project(hector_gazebo_plugins VERSION 0.5.3)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

## Find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(std_srvs REQUIRED)
find_package(gazebo REQUIRED)
find_package(gazebo_ros REQUIRED)
find_package(geographic_msgs REQUIRED)
find_package(geometry_msgs REQUIRED)
find_package(nav_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(tf2 REQUIRED)
find_package(tf2_ros REQUIRED)
find_package(tf2_geometry_msgs REQUIRED)
find_package(rosidl_default_generators REQUIRED)
find_package(rmw_implementation_cmake REQUIRED)

## Find gazebo
if(POLICY CMP0054)
  cmake_policy(SET CMP0054 NEW)
endif()
# find_package(gazebo REQUIRED)
# set(GAZEBO_INCLUDE_DIRS "/usr/include/gazebo-11/gazebo/")
# set(GAZEBO_LIBRARIES "/usr/lib/x86_64-linux-gnu/libgazebo-11.so")
# set(GAZEBO_LIBRARY_DIRS "/usr/lib/x86_64-linux-gnu/")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${GAZEBO_CXX_FLAGS}")
include_directories(${GAZEBO_INCLUDE_DIRS})
link_directories(${GAZEBO_LIBRARY_DIRS})

## Find Boost
# find_package(Boost REQUIRED COMPONENTS thread)
# include_directories(${Boost_INCLUDE_DIRS})

# Set includes
set(INCLUDE_DIRS include 
    ${sensor_msgs_INCLUDE_DIRS}
    ${rclcpp_INCLUDE_DIRS}
    ${CMAKE_CURRENT_BINARY_DIR}/srv
)
include_directories(${INCLUDE_DIRS})

set(LIBS
    geographic_msgs 
    geometry_msgs
    sensor_msgs
    std_msgs
    rclcpp
    tf2
    tf2_geometry_msgs
    tf2_ros
    nav_msgs
    gazebo
    gazebo_ros
)

#######################################
## Declare ROS messages and services ##
#######################################

## Generate messages
rosidl_generate_interfaces(${PROJECT_NAME}
  srv/SetBias.srv
  srv/SetReferenceGeoPose.srv
  DEPENDENCIES geographic_msgs geometry_msgs
)

###########
## Build ##
###########

set(HECTOR_GAZEBO_ROS_PLUGINS_LIST "")

# add_library(diffdrive_plugin_6w src/diffdrive_plugin_6w.cpp)
# target_link_libraries(diffdrive_plugin_6w ${Boost_LIBRARIES} ${GAZEBO_LIBRARIES} ${catkin_LIBRARIES})

# add_library(diffdrive_plugin_multi_wheel src/diffdrive_plugin_multi_wheel.cpp)
# target_link_libraries(diffdrive_plugin_multi_wheel ${Boost_LIBRARIES} ${GAZEBO_LIBRARIES} ${catkin_LIBRARIES})

# add_library(gazebo_ros_force_based_move src/gazebo_ros_force_based_move.cpp)
# target_link_libraries(gazebo_ros_force_based_move ${Boost_LIBRARIES} ${GAZEBO_LIBRARIES} ${catkin_LIBRARIES})

# add_library(hector_gazebo_reset_plugin src/reset_plugin.cpp)
# target_link_libraries(hector_gazebo_reset_plugin ${GAZEBO_LIBRARIES})
# list(APPEND HECTOR_GAZEBO_ROS_PLUGINS_LIST hector_gazebo_reset_plugin)
# ament_target_dependencies(hector_gazebo_reset_plugin ${LIBS})

# add_library(hector_gazebo_ros_imu src/gazebo_ros_imu.cpp)
# target_link_libraries(hector_gazebo_ros_imu ${Boost_LIBRARIES} ${GAZEBO_LIBRARIES})
# list(APPEND HECTOR_GAZEBO_ROS_PLUGINS_LIST hector_gazebo_ros_imu)
# ament_target_dependencies(hector_gazebo_ros_imu ${LIBS})

# add_library(hector_gazebo_ros_magnetic src/gazebo_ros_magnetic.cpp)
# target_link_libraries(hector_gazebo_ros_magnetic ${GAZEBO_LIBRARIES})
# list(APPEND HECTOR_GAZEBO_ROS_PLUGINS_LIST hector_gazebo_ros_magnetic)
# ament_target_dependencies(hector_gazebo_ros_magnetic ${LIBS})

add_library(hector_gazebo_ros_gps src/gazebo_ros_gps.cpp)
target_link_libraries(hector_gazebo_ros_gps ${GAZEBO_LIBRARIES})
# ament_target_dependencies(hector_gazebo_ros_gps ${LIBS})
list(APPEND HECTOR_GAZEBO_ROS_PLUGINS_LIST hector_gazebo_ros_gps)
ament_target_dependencies(hector_gazebo_ros_gps ${LIBS})

# add_library(hector_gazebo_ros_sonar src/gazebo_ros_sonar.cpp)
# target_link_libraries(hector_gazebo_ros_sonar ${GAZEBO_LIBRARIES})
# list(APPEND HECTOR_GAZEBO_ROS_PLUGINS_LIST hector_gazebo_ros_sonar)
# ament_target_dependencies(hector_gazebo_ros_sonar ${LIBS})

# add_library(hector_servo_plugin src/servo_plugin.cpp)
# target_link_libraries(hector_servo_plugin ${Boost_LIBRARIES} ${GAZEBO_LIBRARIES} ${catkin_LIBRARIES})

# Connect targets with locally defined messages
# get_available_rmw_implementations(rmw_implementations2)
# foreach(rmw_implementation ${rmw_implementations2})
#   find_package("${rmw_implementation}" REQUIRED)
#   get_rmw_typesupport(typesupport_impls "${rmw_implementation}" LANGUAGE "cpp")
#   foreach(typesupport_impl ${typesupport_impls})
#     rosidl_target_interfaces(hector_gazebo_ros_imu ${PROJECT_NAME} ${typesupport_impl})
#     # message(WARNING, ${typesupport_impl})
#     rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} ${typesupport_impl})
#   endforeach()
# endforeach()
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_cpp)
rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_introspection_cpp)
rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_fastrtps_cpp)
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_introspection_cpp)
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_service_type_support_tv) 
# #############
# ## Install ##
# #############

install(DIRECTORY include/${PROJECT_NAME}/
  DESTINATION include
  FILES_MATCHING PATTERN "*.h"
  PATTERN ".svn" EXCLUDE
)

install(TARGETS ${HECTOR_GAZEBO_ROS_PLUGINS_LIST}
  DESTINATION lib/${PROJECT_NAME}
  LIBRARY DESTINATION lib
)

# install(TARGETS
#   diffdrive_plugin_6w
#   diffdrive_plugin_multi_wheel
#   gazebo_ros_force_based_move
#   hector_gazebo_reset_plugin
#   hector_gazebo_ros_imu
#   hector_gazebo_ros_magnetic
#   hector_gazebo_ros_gps
#   hector_gazebo_ros_sonar
#   hector_servo_plugin
#   ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
#   LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
#   RUNTIME DESTINATION ${CATKIN_GLOBAL_BIN_DESTINATION}
# )

ament_package()

```
If a solution can be found I can help with a pull requst to others can use it in Humble as well.

Martin-Oehler, 2022-08-04:
This seems to be the same problem as #93.

austin-InDro, 2022-08-04:
I've looked at https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/issues/93 but their issue seemed to pertain to CMake outputting .a files, not .so. However, I'm able to import .so files

cadmus-to, 2022-08-06:
Maybe silly question, but is that the output of Gazebo using ROS 1? I haven't tried using ROS 1 and ROS 2 before but are you sure that works?

austin-InDro, 2022-08-08:
I'm most likely doing something silly so I appreciate the questions! No, it's definately ROS2 since I haven't installed ROS1 on this system. Is it manditory that there is geometry associated with the link before passing it into the GPS plugin?

Martin-Oehler, 2022-08-08:
At least in Gazebo (Classic), links without associated physical properties (mass, inertia) are removed, unless you mark them with a special tag.

austin-InDro, 2022-08-08:
That wouldn't case the plugin load error though, would it?

Wimll, 2022-10-17:
> So I was able to load the plugin by just moving the compiled library to the gazebo pluigin directory instead of sourcing my own build directory.
> 
> However, another error has appeared
> 
> ```
> [gazebo-1] gzserver: symbol lookup error: /usr/lib/x86_64-linux-gnu/gazebo-11/plugins/libhector_gazebo_ros_gps.so: undefined symbol: _ZN22rosidl_typesupport_cpp31get_service_type_support_handleIN21hector_gazebo_plugins3srv19SetReferenceGeoPoseEEEPK29rosidl_service_type_support_tv
> ```

Hi @austin-InDro , did you ever manage to solve this error?

austin-InDro, 2022-10-17:
Hi @Wimll, yeah. I moved it to the plugin directory with all other Gazebo Plugins manually. The error went away after that.

Wimll, 2022-10-17:
Hmm thanks, still sounds like this all needs some proper reworking to actually function on Humble

austin-InDro, 2022-10-17:
Yeah... I'll close this issue, but the resolution isn't ideal haha. When I have a few moments over the next few days. I can open another ticket with the current error I'm having. If we can fix that, the plugins will work on Humble.

closed 2022-10-17 by austin-InDro

Wimll, 2022-10-18:
Hi @austin-InDro , I solved the linking errors by adding the following lines and commenting the existing rosidl_target_interfaces commands:

```# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_cpp)
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_introspection_cpp)
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_fastrtps_cpp)
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_typesupport_introspection_cpp)
# rosidl_target_interfaces(hector_gazebo_ros_gps ${PROJECT_NAME} rosidl_service_type_support_tv) 

rosidl_get_typesupport_target(IDL_TARGETS ${PROJECT_NAME} rosidl_typesupport_cpp)
target_link_libraries(hector_gazebo_ros_gps ${IDL_TARGETS})```

I had to dive into the sourcecode of rosidl to fix this, but it seems to work.
The plugin now works fine, without copying it to the default gazebo plugins folder.
I get GPS output, I did not add a fix for the IMU as I am using the default IMU sensor simulator from Gazebo.
What errors were you getting after this?

austin-InDro, 2022-10-26:
```
In file included from /home/austingreisman/Documents/ws/src/hector_gazebo/hector_gazebo_plugins/src/gazebo_ros_gps.cpp:29:
/home/austingreisman/Documents/ws/src/hector_gazebo/hector_gazebo_plugins/include/hector_gazebo_plugins/gazebo_ros_gps.h:40:10: fatal error: hector_gazebo_plugins/srv/set_reference_geo_pose.hpp: No such file or directory
   40 | #include <hector_gazebo_plugins/srv/set_reference_geo_pose.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/hector_gazebo_ros_gps.dir/build.make:76: CMakeFiles/hector_gazebo_ros_gps.dir/src/gazebo_ros_gps.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:678: CMakeFiles/hector_gazebo_ros_gps.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
In file included from /home/austingreisman/Documents/ws/src/hector_gazebo/hector_gazebo_plugins/src/gazebo_ros_imu.cpp:29:
/home/austingreisman/Documents/ws/src/hector_gazebo/hector_gazebo_plugins/include/hector_gazebo_plugins/gazebo_ros_imu.h:48:10: fatal error: hector_gazebo_plugins/srv/set_bias.hpp: No such file or directory
   48 | #include <hector_gazebo_plugins/srv/set_bias.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/hector_gazebo_ros_imu.dir/build.make:76: CMakeFiles/hector_gazebo_ros_imu.dir/src/gazebo_ros_imu.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:626: CMakeFiles/hector_gazebo_ros_imu.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```