tu-darmstadt-ros-pkg/hector_gazebo #99: Inconsistent initial orientation between melodic/gazebo9 and noetic/gazebo11

lkm1321, 2023-05-18:
Hi,

I am finding that the initial orientation of the IMU is inconsistent between melodic/gazebo9 and noetic/gazebo11 by 90 degrees. 
Without any motion applied to the robot/IMU, melodic/gazebo9 outputs initial orientation (x, y, z, w) = (0, 0, 0, 1), and noetic/gazebo11 (x, y, z, w) = (0, 0, -0.707, 0.707)

We are using the following snippet:

```
  <gazebo>
    <plugin name="imu_controller" filename="libhector_gazebo_ros_imu.so">
      <robotNamespace>$(arg robot_namespace)</robotNamespace>
      <updateRate>40.0</updateRate>
      <bodyName>imu_enu</bodyName>
      <frameId>imu_enu</frameId>      
      <topicName>an_device/Imu</topicName>
      <accelDrift>0.005 0.005 0.005</accelDrift>
      <accelGaussianNoise>0.005 0.005 0.005</accelGaussianNoise>
      <rateDrift>0.005 0.005 0.005 </rateDrift>
      <rateGaussianNoise>0.005 0.005 0.005 </rateGaussianNoise>
      <headingDrift>0.005</headingDrift>
      <headingGaussianNoise>0.005</headingGaussianNoise>
    </plugin>
  </gazebo>
```

lkm1321, 2023-05-19:
Thanks for getting back to me. To confirm: 

1. Is world is in ENU?
2. With rpyOffset = 0, 0, 0, should the orientation match between the IMU readings and the link specified in bodyName? In this case, melodic/gazebo11 may be in fault. Can test and report back

lkm1321, 2023-05-19:
Thought I'd add more background to 2: our URDF model does have an offset of 90 degrees yaw between base_link and IMU

Martin-Oehler, 2023-05-19:
> Is world in ENU?

Yes.

> Thought I'd add more background to 2: our URDF model does have an offset of 90 degrees yaw between base_link and IMU

This means, the IMU is mounted with an offset, so the values in gazebo11 are probably correct and gazebo9 ignores the mounting offset.